bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #169: cannot flash firmware.bin and cannot update bootloader?

spikar85, 2024-08-24:
I’m reaching out for assistance with my BTT Octopus Board. I’ve been encountering some issues flashing the firmware. 

Specifically:

When attempting to flash firmware.bin via the SD card, the file does not seem to change to FIRMWARE.CUR.

A suggestion from an online forum indicated that this might be related to the bootloader.

I followed the steps outlined in the DFU update guide to update the bootloader (link: [DFU Update Bootloader Guide](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/blob/master/Firmware/DFU%20Update%20bootloader/Octopus-DFU%20Update%20bootloader%20-%20en.pdf)). However, I am now unable to establish a connection via USB-C. Despite trying multiple USB cables and ports, I keep receiving a "No DFU Detected" error.

Could you provide guidance on the following:

• Any troubleshooting steps for the "No DFU Detected" issue?
• Possible reasons why the firmware is not switching to FIRMWARE.CUR?
[image]

takemeplaces, 2025-01-08:
I have had the exact same issue twice now. I started with an F446 and now a H7 pro. Was able to flash the bootloader from Bigtreetechs Github for this. still showed usb port correctly. flashed firmware via sd card. nothing changed and no change from Bin to CUR. flashed via STM Programmer and now am unable to comm to the card entirely.

davidsonrsouza, 2025-10-15:
Também estou com o mesmo problema. Ao conectar o microSD não atualizava o firmware. Então segui o passo a passo para reinstalar o bootloader. No qual todas as etapas foram concluídas com sucesso.
Apenas a última etapa descrita no passo a passo, que é pressionar o botão reset eu não fiz. 
A placa ficou ligada por algumas horas após essa reinstalação do bootloader. Depois simplesmente eu a  desliguei para retirar o jumper e a mesma não ligou mais ao retornar com a alimentação.
Achei estranho que até os LEDs das alimentações não estão se acendendo mais.

No meu caso, se trata da placa Octopus Pro V1.0 e todos esses procedimentos foram realizados via alimentação USB-C.