bigtreetech/BIGTREETECH-OCTOPUS-V1.0 #176: Minimal firmware.bin using platformIO for Octopus 1.1

unrealwill, 2025-03-23:
I succeeded to flash Marlin on my octopus 1.1 board, by copying the generated firmware.bin to the SD card, where it automatically gets renamed to FIRMWARE.cur and it works with pronterface.

I am new to platformIO.

My board is an Octopus 1.1, with a STM32F446ZET6 chip.

I am now trying to blink the on board LED, which according to the schematic is on PA_13.

I tried https://github.com/maxgerhardt/nucleo-f446ze-with-arduino/tree/main and selecting octopus or octopus_min environment in platformIO environment selector. 

It generates a firmware.bin successfully, which I copy to the SD card where it automatically gets renamed to FIRMWARE.cur, so I think the flashing is occurring successfully, but the LED doesn't blink.

**Could you to create a minimal boilerplate so that people can use their board with custom firmware more easily ?**

I spent all afternoon yesterday battling with build flags, and configuration, doing SD uploads, but without success.

I have not yet tried the DFU upload procedure, and would rather not, as I don't want to have to play with jumpers for every upload, and don't want to brick my board which can currently load Marlin.

Thanks

PS:
I have searched the internet a lot to solve this issue, and I'm not the only one facing this issue. I can't understand the logic of selling an open-source board, but not providing the bare minimum to get started with custom code, please forgive me I must be missing something.

futurexdesign, 2025-03-31:
@unrealwill Please try this: https://github.com/futurexdesign/Octopus_Minimal 

I have been using this board for awhile with platform.io, so i put this together to hopefully get you bootstrapped.   There was a bunch of odd stuff around the custom variant and board definition to get it to work correctly with pio.  That stuff should all be in that sample project for you. 

It will build a firmware.bin that will work with the stock BTT supplied boot loader if you but it on a card and boot it.

unrealwill, 2025-03-31:
@futurexdesign Thanks a lot ! 👍 👍 👍 

I had to fix three minor problems to your code to make it work : 

This line  ` pinMode(LED,` FAN_OUTPUT);` was wrong (see https://www.arduino.cc/reference/fr/language/functions/digital-io/pinmode/ ), which was making the code crashed. But the device was still detected on USB.

The LEDPIN on my device is PA13 and not PD13.

I also added a counter to check that the delay function wait the correct time, and it works fine :) counting to 50 in 25s.

Add to platformio.ini : 
`monitor_speed = 115200`

```
#include <Arduino.h>
#define LEDPIN                            PA13

bool LED = false;

void setup() {
    Serial.begin(115200);
    pinMode(LEDPIN, OUTPUT);
    Serial.println("Hello From Octopus");
}
int co = 0;

void loop() {

    Serial.println("Hello From Octopus");
    Serial.println( co );
    co++;
    digitalWrite(LEDPIN, LED);
    LED = !LED;

    delay(500);
}
```

futurexdesign, 2025-03-31:
Glad it helped.  I through that together in like 3 minutes, sorry I goofed it up.  Could you make an MR back to the example repo?

Good luck with your project!